tdorssers/TeslaPy #149: Selenium blocked by Tesla

Coverttrickster, 2023-11-08:
As of yesterday I attempted a login with selenium and TeslaPy and tesla is able to detect the use of the selenium. I attempted to add options that help prevent the detection of selenium such as:
`
chrome_options.add_argument('--disable-blink-features=AutomationControlled')
chrome_options.add_experimental_option("excludeSwitches", ["enable-automation"])
chrome_options.add_experimental_option('useAutomationExtension', False)
`

I also tried to rotate user agents as such:
`
driver.execute_cdp_cmd('Network.setUserAgentOverride', {"userAgent": 'TeslaApp/4.10.0 Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/83.0.4103.53 Safari/537.36'})
`
but no luck.

Here is the page I get redirected to when I disable headless mode. I attempted to complete all of the tasks by hand but it loops around the challenges with no luck of proving I am human.
[image]

Are there any other methods I could attempt to retrieve the refresh token without having an end user do anything more than type in the login credentials?

Coverttrickster, 2023-11-11:
With the release of the official API my guess is Tesla has decided to further increase security and prevent non-official access to the API.

RichieB2B, 2023-11-11:
That makes sense, see #150