tdorssers/TeslaPy #150: Move to Fleet API and Tesla Vehicle Command SDK

RichieB2B, 2023-11-11:
As [announced](https://developer.tesla.com/docs/fleet-api#announcements-amp-api-changelog) by [Tesla](https://github.com/teslamotors/vehicle-command), the Owner API will stop working "starting 2024". TeslaPy will need to move to using the Fleet API (for pre-2021 cars) and Tesla Vehicle Command SDK (for post-2021 cars).

goldhoorn, 2023-11-29:
This is a critial issue, i am getting already the following erro:
`requests.exceptions.HTTPError: 403 Client Error: Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning for url: https://owner-api.teslamotors.com/api/1/vehicles/1492932313807820/command/charge_stop
`

However the deprication is set for next year TeslaPy stopped working for me :/

fabianhu, 2023-12-17:
Hi, this may be seen as a blatant self-advertising, as it is: 
I just published my Python implementation of the new Tesla API here:
[tesla-api](https://github.com/fabianhu/tesla_api)

So far it works for me including the whole registration process. If someone wants to hop in, you are invited. As the structure is much simpler as TeslaPy, a pull request does not work so far.

For producing the signed commands I am using the tesla-command in written in go by tesla, which can also be compiled for Raspberry Pi.
A python native implementation can be done, when the documentation from Tesla is published.

In the past I used TeslaPy, which worked great!
If someone feels offended by this post, please let me know, I will delete/edit it asap.
@tdorssers

JanJaapKo, 2023-12-30:
@tdorssers is this something you're (planning to be) working on? If so, any help needed?

I've made a Domoticz plugin around this library (which works really well I must say) so curious to know if and how to update when implementing the new Tesla API.

RichieB2B, 2024-01-20:
I just did the work for [TWCManager](https://github.com/ngardiner/TWCManager/pull/546). In short these were the required steps:

1. Register an application with Tesla and your cars
2. Get an OAuth token for each car (yes, there now is an official way to do this)
3. Set up the proxy [supplied](https://github.com/teslamotors/vehicle-command) by Tesla for your app
4. Make the API server that TeslaPy uses configurable. This can be a FleetAPI region server or a proxy URL.
5. Don't use the car API ID in the command URLs. The proxy only supports the VIN in command URLs.

I'm not sure step 5 is needed for TeslaPy, maybe it already uses the VIN.