tdorssers/TeslaPy #154: Not able to login

tomdddd, 2024-01-07:
Hi all,

I am quite a noob in programming but was using the teslapy library for quite a while, until it did not work anymore.
I have been searching to get it solved but cannot find it. The scripts are running on a raspberryPI4.
Could someone help out to solve the login failure?
When I run the script I do not get any browser popup to login in again via auth.tesla.com...

[image]

Haselmaier, 2024-01-15:
Have you made any progress on this?  

I'm experiencing the same thing.  The vehicle_list call is failing.  I established new auth tokens and it behaves the same way.  The code I'm running has been working well for well over a year.

I fear Tesla is locking down access to this (unsupported) API.

tomdddd, 2024-01-15:
Hi HaselMaier,

No, not any progress yet... Do you know if "supported' API are available?

Haselmaier, 2024-01-15:
I only know of this one API.  I read somewhere it's access and use is all without the endorsement of Tesla....that someone or group figured out the auth stuff and reverse engineered what could be gotten from it.  Then Tim put this nice Python layer on top.

tomdddd, 2024-01-15:
Did someone already tested this?
https://developer.tesla.com/docs/fleet-api?python#charge_history

mababio, 2024-01-16:
I'm not facing any issues. Well at least not yet.

mababio, 2024-01-23:
facing issue with API now.... reported in #156