tdorssers/TeslaPy #155: It seems get_charge_history() API URI is no longer working? Returns 400

nohir0, 2024-01-21:
teslapy==2.8.0

Trying:
`print(vehicles[0].get_charge_history())`

throws:
`
Traceback (most recent call last):
[...]
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: https://powergate.prd.sn.tesla.services:443/api/v4/vehicle/charge_history => {:Message=>"Error getting charge history", :Status=>400} for url: https://owner-api.teslamotors.com/api/1/vehicles/[my-tesla-id]/charge_history
`

A colleage uses this in his script for fetching charging history now:
`
url_charging_base = 'https://ownership.tesla.com/mobile-app/charging/'
url_charging_history = f'{url_charging_base}history?deviceLanguage=en&deviceCountry=AT&httpLocale=en_US&vin={vehicle_vin}&operationName=getChargingHistoryV2'
`

What do you mean?

tdorssers, 2024-01-27:
Can you checkout commit e9d4612d68bdadbd2618e6c30229a2e81087d9a0 to see if this new method works for you?

raythefourth, 2024-04-06:
The URL works (https://ownership.tesla.com/mobile-app/charging/history) and it provides great detail.  It only provides data for public charger sessions however. No data is provided for home charging sessions with my Tesla Wall Charger.

Here's an example of the returned data:
`
"data": [
        {
            "sessionId": [redacted],
            "vin": "[redacted]",
            "postId": "0",
            "cabinetId": "SC000000000008927",
            "din": "",
            "chargeSessionId": "[redacted]",
            "siteLocationName": "Buellton, CA",
            "siteAmenities": [
                "AMENITIES_WIFI",
                "AMENITIES_RESTROOMS",
                "AMENITIES_CAFE",
                "AMENITIES_LODGING",
                "AMENITIES_RESTAURANT",
                "AMENITIES_TWENTY_FOUR_HOUR"
            ],
            "siteEntryLocation": {
                "latitude": 34.614548,
                "longitude": -120.188179
            },
            "siteAddress": {
                "street": "McMurray Rd",
                "streetNumber": "555",
                "city": "Buellton",
                "district": null,
                "state": "CA",
                "countryCode": "US",
                "country": "United States",
                "postalCode": "93427"
            },
            "siteLocalizedName": "NA-US-CA-Buelton",
            "chargeStartDateTime": "2023-11-13T16:34:22-08:00",
            "chargeStopDateTime": "2023-11-13T16:43:21-08:00",
            "unlatchDateTime": "2023-11-13T16:43:23-08:00",
            "countryCode": "US",
            "isDcEnforced": false,
            "isMsp": false,
            "trtId": 2161,
            "terminalSerialNumber": null,
            "sessionSource": null,
            "isV4Session": false,
            "credit": null,
            "chargingPackage": null,
            "chargingVoucher": null,
            "disputeDetails": null,
            "fees": [
                {
                    "sessionFeeId": [redacted],
                    "feeType": "CHARGING",
                    "currencyCode": "USD",
                    "pricingType": "PAYMENT",
                    "rateBase": 0.39,
                    "rateTier1": 0,
                    "rateTier2": 0,
                    "rateTier3": null,
                    "rateTier4": null,
                    "usageBase": 7,
                    "usageTier1": 8,
                    "usageTier2": 0,
                    "usageTier3": null,
                    "usageTier4": null,
                    "totalBase": 2.73,
                    "totalTier1": 0,
                    "totalTier2": 0,
                    "totalTier3": 0,
                    "totalTier4": 0,
                    "totalDue": 2.73,
                    "voucherValue": null,
                    "netDue": 2.73,
                    "uom": "kwh",
                    "isPaid": true,
                    "status": "PAID",
                    "processFlagId": 1
                },
                {
                    "sessionFeeId": [redacted],
                    "feeType": "PARKING",
                    "currencyCode": "USD",
                    "pricingType": "NO_CHARGE",
                    "rateBase": 0,
                    "rateTier1": 0,
                    "rateTier2": 0,
                    "rateTier3": null,
                    "rateTier4": null,
                    "usageBase": 0,
                    "usageTier1": 0,
                    "usageTier2": 0,
                    "usageTier3": null,
                    "usageTier4": null,
                    "totalBase": 0,
                    "totalTier1": 0,
                    "totalTier2": 0,
                    "totalTier3": 0,
                    "totalTier4": 0,
                    "totalDue": 0,
                    "voucherValue": null,
                    "netDue": 0,
                    "uom": "min",
                    "isPaid": true,
                    "status": "PAID",
                    "processFlagId": 1
                }
            ],
            "billingType": "IMMEDIATE",
            "invoices": [
                {
                    "fileName": "[redacted].pdf",
                    "contentId": "[redacted]",
                    "invoiceType": "IMMEDIATE",
                    "beInvoiceId": "[redacted]"
                }
            ],
            "fapiaoDetails": null,
            "programType": "PTSCH",
            "vehicleMakeType": "TSLA",
            "surveyCompleted": false,
            "surveyType": "CSAT",
            "chargingSiteType": "",
            "operator": null,
            "suboperator": null,
            "owner": null
        }
`