tdorssers/TeslaPy #156: Seems like the teslapy is no longer working with the new Tesla APIs

aviadoffer, 2024-01-23:
I'm now getting an error : Endpoint is only available on fleetapi
I understand that this is due to the new API requirements, just wondering if there is a plan to update Teslapy project?
Thanks in advance!

israndy, 2024-01-23:
Perhaps the intent was to rely on Tesla to support these apps, sadly I made my code rather reliant on the easy calls in TeslaPy

doubledrat, 2024-01-23:
Same here on vehicle_list()

requests.exceptions.HTTPError: 412 Client Error: Endpoint is only available on fleetapi. Visit https://developer.tesla.com/docs for more info for url: https://owner-api.teslamotors.com/api/1/vehicles

DaveTBlake, 2024-01-23:
Yes access to the https://owner-api.teslamotors.com/api/1/vehicles endpoint stopped sometime over night (in UK), I would guess that for vehicle data you now have to use the Fleet API.

However `api/1/products` still works, so if you are using TeslaPy for powerwalls then removing refernce to `tesla.vehicle_list()` from your client scripts will keep working for now.

doubledrat, 2024-01-23:
> Yes access to the https://owner-api.teslamotors.com/api/1/vehicles endpoint stopped sometime over night (in UK), I would guess that for vehicle data you now have to use the Fleet API.
>
> However [...]

yes, but for how long? :(

I looked at registering for the fleet api, but you have to be a business :(

Chillywasher, 2024-01-23:
In the same boat as you guys. I just tried the Home Assistant Tesla HACS integration and that works fine for 2 cars so I'm guessing there other implementations available to get at our Tesla data.

weljajoh, 2024-01-23:
This patch makes it work again for me by switch from VEHICLE_LIST to PRODUCT_LIST and filtering out batteries and solar (there is no 'vehicle' resource_type in the vehicle product response)

```
--- __init__.py.orig	2024-01-23 10:27:36.591958142 +0100
+++ __init__.py	2024-01-23 11:25:29.264901231 +0100
@@ -372,7 +372,8 @@
 
     def vehicle_list(self):
         """ Returns a list of `Vehicle` objects """
-        return [Vehicle(v, self) for v in self.api('VEHICLE_LIST')['response']]
+        return [Vehicle(v, self) for v in self.api('PRODUCT_LIST')['response']
+                if v.get('resource_type') != 'battery' and v.get('resource_type') != 'solar']
 
     def battery_list(self):
         """ Returns a list of `Battery` objects """
```

aviadoffer, 2024-01-23:
I was able to register my app on tesla dev site. so now I have the keys from tesla. but how do I plug those into this API ?

dan-cristian, 2024-01-24:
Same issue here. Looking at the doc there seems to be a way to make it work, for free, for a while, with limited no. of requests. https://developer.tesla.com/docs/fleet-api

dan-cristian, 2024-01-24:
I tried to create a business account using valid business details but got the error at the end: Unable to Onboard, Contact your account manager....

aviadoffer, 2024-01-24:
@dan-cristian  uncheck the first option (get personal data etc) and it will work.

aviadoffer, 2024-01-24:
For the record I used https://github.com/fabianhu/tesla_api and manage to get everything I need.
this method worked for me after help from the author

DeLN0, 2024-01-24:
> This patch makes it work again for me by switching from VEHICLE_LIST to PRODUCT_LIST and filtering out batteries and solar (there is no 'vehicle' resource_type in the vehicle product response)
> 
> ```
> --- __init__.py.orig	2024-01-23 10:27:36.591958142 +0100
> +++ __init__.py	2024-01-23 11:25:29.264901231 +0100
> @@ -372,7 +372,8 @@
>  
>      def vehicle_list(self):
>          """ Returns a list of `Vehicle` objects """
> -        return [Vehicle(v, self) for v in self.api('VEHICLE_LIST')['response']]
> +        return [Vehicle(v, self) for v in self.api('PRODUCT_LIST')['response']
> +                if v.get('resource_type') != 'battery' and v.get('resource_type') != 'solar']
>  
>      def battery_list(self):
>          """ Returns a list of `Battery` objects """
> ```

This works as long as there are no other products than battery and solar.
I have a Powerwall and this crashes.

A better way is to check if product has vin and vehicle_id.
Use the following code and it works:

-        if v.get('resource_type') != 'battery' and v.get('resource_type') != 'solar']     
+        return [Vehicle(v, self) for v in self.api('PRODUCT_LIST')['response'] 
+               if 'vin' in v and 'vehicle_id' in v]

DeLN0, 2024-01-24:
> This patch makes it work again for me by switching from VEHICLE_LIST to PRODUCT_LIST and filtering out batteries and solar (there is no 'vehicle' resource_type in the vehicle product response)
>
> ``` [...]

This works as long as there are no other products than battery and solar.
I have a Powerwall and this crashes.

A better way is to check if product has vin and vehicle_id.
Use the following code and it works:

    def vehicle_list(self):
        """ Returns a list of `Vehicle` objects """  
        return [Vehicle(v, self) for v in self.api('PRODUCT_LIST')['response'] 
                if 'vin' in v and 'vehicle_id' in v]

DaveTBlake, 2024-01-24:
I just tried to create a business account using valid UK business details (by tax id it wants a VAT number and not a UTR)  but got the error at the end: Unable to Onboard, Contact your account manager.

The only scope I checked was Powerwall Information, the first option get personal data was unchecked. Suggestions?

DeLN0, 2024-01-24:
> I just tried to create a business account using valid UK business details (by tax id it wants a VAT number and not a UTR) but got the error at the end: Unable to Onboard, Contact your account [...]

As a suggestion try to enter browser console and check what messages you have there.

DeLN0, 2024-01-24:
> @dan-cristian uncheck the first option (get personal data etc) and it will work.

I also have this problem.
In my case, i can't uncheck the first option because i can't go back to the first step.

Any suggestions?

[image]

DaveTBlake, 2024-01-24:
@DeLN0 I know that for now the Owner API works for Powerwall settings, and I have modified TeslaPy to keep my script working. But the end of Owner API is coming so I'm looking to see if I can jump throught the hoops to move the FleetAPI 

> As a suggestion try to enter browser console and check what messages you have there.

Neat idea. Console shows 
`Content Security Policy: The page's settings blocked the loading of a resource at eval ("script-src").`

Not sure what to make of that.  My guess is that my URLs are the problem, I own a domain but use free-parking and that doesn't have a certificate. I couldn't see how to get Localhost working either, which could be all I need. My next step is to look for some free storage I have point my domain at. All in a lot of effort just to get a script for one user working.

DeLN0, 2024-01-24:
@DaveTBlake

> Not sure what to make of that. My guess is that my URLs are the problem, I own a domain but use free-parking and that doesn't have a certificate. I couldn't see how to get Localhost working either, [...]

In my case, if i go to https://developer.tesla.com on my iPhone this is what i get, even though my domain is fine and certicates are up to date:

[image]

88gts, 2024-01-24:
> @DaveTBlake
>
>
> In my case, if i go to https://developer.tesla.com on my iPhone this is what i get, even though my domain is fine and certicates are up to date:

i followed your recommendations and overrode the vehicle_list function but i am now seeing issues with my local cert : HTTPError: 429 Client Error: Too Many Requests for url: https://owner-api.teslamotors.com/api/1/vehicles/374XXXXXX

i also have my own domain but lack a cert. let me know what you find.

mshoe007, 2024-01-25:
I managed to get developer access working yesterday following this tutorial.

https://shankar-k.medium.com/tesla-developer-api-guide-account-setup-app-creation-registration-and-third-party-da24aba1bddd

israndy, 2024-01-25:
Paywalled

mshoe007, 2024-01-25:
https://12ft.io/https://shankar-k.medium.com/tesla-developer-api-guide-account-setup-app-creation-registration-and-third-party-da24aba1bddd

DaveTBlake, 2024-01-25:
@israndy that tutorial by Shanka Kumarasamy is also just in his blog too (not behind the Medium paywall) https://shankarkumarasamy.blog/2023/10/29/tesla-developer-api-guide-account-setup-app-creation-registration-and-third-party-authentication-configuration-part-1/

the-mace, 2024-01-25:
> > I just tried to create a business account using valid UK business details (by tax id it wants a VAT number and not a UTR) but got the error at the end: Unable to Onboard, Contact your account manager.
> > The only scope I checked was Powerwall Information, the first option get personal data was unchecked. Suggestions?
> 
> [DaveTBlake](https://github.com/DaveTBlake) you can use the old API, you just need to replace def vehicle_list(self) function in the `__init__.py` file from teslapy library with this function:
> 
> ```
> def vehicle_list(self):
>     """ Returns a list of `Vehicle` objects """  
>     return [Vehicle(v, self) for v in self.api('PRODUCT_LIST')['response'] 
>             if 'vin' in v and 'vehicle_id' in v]
> ```

This would be good to to have in teslapy as a PR. This change worked for me, thanks!

mshoe007, 2024-01-25:
That patch allows TeslaPy to continue to do 'read' sorts of functions. But, if I try to issue a command, I get

```
 Client Error: Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api
```

I think the old commands API now only works for pre refresh (2021) S & X?

The rest of us have to use the Fleet API for sending commands to the car.

aviadoffer, 2024-01-25:
@mshoe007  I have a 2020 X and a 2022 X, both works fine with new API's to read data and to run commands. so moving to the fleet API works for me on both cars

JanJaapKo, 2024-01-25:
@aviadoffer I gave the https://github.com/fabianhu/tesla_api a try but got stuck at the step of building the go language part. Are there any extensions to be made to the readme to make it clearer?

aviadoffer, 2024-01-25:
@JanJaapKo  so I only did the simple step of importing the tesla_api_2024.py for the purpose of auth. placing the keys in TeslaKeys

so in a high level it looks like this:

from tesla_api import tesla_api_2024, TeslaAPI
teslaApi = TeslaAPI()
vehicles = teslaApi.get_vehicles_list()

Then I added the function I needed to the tesla_api_2024, example:
 def tesla_start_charge(self, vin):

        conn = http.client.HTTPSConnection("fleet-api.prd.na.vn.cloud.tesla.com")
        payload = json.dumps({})
        headers = {
            'Content-Type': 'application/json',
            'Authorization': f'Bearer {self.access_token}'
        }
        conn.request("POST", f"/api/1/vehicles/{vin}/command/charge_start", payload, headers)
        res = conn.getresponse()
        data = res.read()

and 
 def tesla_set_charge_amp(self, vin , amps):

        conn = http.client.HTTPSConnection("fleet-api.prd.na.vn.cloud.tesla.com")
        payload = json.dumps({
            "charging_amps": f"{amps}"
        })
        headers = {
            'Content-Type': 'application/json',
            'Authorization': f'Bearer {self.access_token}'
        }
        conn.request("POST", f"/api/1/vehicles/{vin}/command/set_charging_amps", payload, headers)
        res = conn.getresponse()
        data = res.read()

etc...

DeLN0, 2024-01-25:
> That patch allows TeslaPy to continue to do 'read' sorts of functions. But, if I try to issue a command, I get
>
> ```
>  Client Error: Tesla Vehicle Command Protocol required, please refer to the [...]

Is your car newer than november 2023?

Tesla says 
`
November 2023	Newly delivered vehicles* will only support the Tesla Vehicle Command protocol after this date
`
More info here: [https://developer.tesla.com/docs/fleet-api#2023-11-17-vehicle-commands-endpoint-deprecation-timeline-action-required](url)

mshoe007, 2024-01-25:
> got stuck at the step of building the go language part. Are there any extensions to be made to the readme to make it clearer? I'm running on a Raspberry Pi 3 btw

The build instructions are here: https://github.com/teslamotors/vehicle-command

You need a 32-bit install of go for a Raspberry Pi 3. (I assume you're running a 32-bit OS on it). If you are running 64-bit linux, get the [go1.21.6.linux-arm64.tar.gz](https://go.dev/dl/go1.21.6.linux-arm64.tar.gz) file.

```
 cd /tmp ; wget https://go.dev/dl/go1.21.6.linux-armv6l.tar.gz
sudo rm -rf /usr/local/go && sudo tar -C /usr/local -xzf go1.21.6.linux-armv6l.tar.gz
PATH="/usr/local/go/bin:$HOME/go/bin:$PATH"
cd
git clone https://github.com/teslamotors/vehicle-command
cd vehicle-command/
go get ./...
go build ./...
go install ./...
ls $HOME/go/bin
ble*               tesla-control*     tesla-keygen*
tesla-auth-token*  tesla-http-proxy*  unlock*
tesla-control --help
Usage: tesla-control [OPTION...] COMMAND [ARG...]

Run tesla-control help COMMAND for more information. Valid COMMANDs are listed below.

 * Commands sent to a vehicle over the internet require a VIN and a token.
 * Commands sent to a vehicle over BLE require a VIN.
 * Account-management commands require a token.

Available OPTIONs:
  -ble
        Force BLE connection even if OAuth environment variables are defined
<etc>
```

Note you need to permanently set your PATH to include $HOME/bin/go if you have not already

fabianhu, 2024-01-26:
> @aviadoffer I gave the https://github.com/fabianhu/tesla_api a try but got stuck at the step of building the go language part. Are there any extensions to be made to the readme to make it clearer? I'm running on a Raspberry Pi 3 btw

please open an issue in my repo, I'll help!

leepfrog-ger, 2024-01-27:
@DaveTBlake
@DeLN0

Is your domain pointing to a CNAME record by any chance? I ran into the same issue, after pointing it directly to an A record it worked

DeLN0, 2024-01-27:
> @DaveTBlake @DeLN0
> 
> Is your domain pointing to a CNAME record by any chance? I ran into the same issue, after pointing it directly to an A record it worked

@leepfrog-ger 
This was exactly my case!
This issue drove me SO mad that i ended up registering a new domain, to which by chance i didn't attach any CNAME record. This new domain was ok with Tesla so the app was approved instantly.

Ty for pointing this out!

DeLN0, 2024-01-27:
@tdorssers 
Ty for the commits!

Do you think the Owner API hence TeslaPy will still work in the future (regardless of what Tesla says) ?
Because for now the Android app still uses Owner API i think.

Are you planning to update TeslaPy to support the new Tesla Vehicle Command protocol?
It seems Tesla makes it possible to set up a proxy server to maintain compatibility with Owner API: 

[https://github.com/teslamotors/vehicle-command](url)

`
Some developers may be familiar with Tesla's Owner API. Owner API will stop working as vehicles begin requiring end-to-end command authentication. If you are one of these developers, you can set up the proxy server or refactor your application to use this library directly. 
`

Ty for your contributions btw!

JanJaapKo, 2024-01-27:
@tdorssers I manually changed the code to fetch the product list iso vehicle list, so it works again. However, when running a new install or upgrade with pip, the version stays at 2.8.0. Any ideas?

Further, this issue seems to mix the broken owner api (which should be fixed with recent commits) and the move to the fleet API to be able to send commands, correct?

mshoe007, 2024-01-27:
@DeLN0 - "Do you think the Owner API hence TeslaPy will still work in the future (regardless of what Tesla says) ?"

Tesla has explicitly said the Owner API will continue to be available for pre 2021 S & X as those vehicles do not support the FleetAPI.  https://developer.tesla.com/docs/fleet-api#2023-11-17-vehicle-commands-endpoint-deprecation-timeline-action-required

> January 2024: All vehicles* will require Tesla Vehicle Command protocol. The REST API will be fully deprecated
where where * is

> *Fleet accounts are excluded from these changes until further notice. **Pre-2021 Model S/X are excluded from these changes**.

Plus, for non-commands, the "REST API" (as Tesla calls it, the "Owner API" as we call it) is not going away for any vehicles.  See the 2023-10-09 Announcement on the fleet-api page (above)

> Following the release of [Tesla Vehicle Command SDK](https://developer.tesla.com/docs/fleet-api#2023-10-09-tesla-vehicle-command-sdk) support for [REST API vehicle command endpoints](https://developer.tesla.com/docs/fleet-api#vehicle-commands) is now reaching end of life. Starting 2024 most vehicles will require sending commands via Tesla Vehicle Command SDK.

The RestAPI/OwnerAPI no longer supports the 'command' parts of the API for 3/y and 2021+ S/X. (with whatever exception "Fleet accounts" means)

@DeLN0 : "Are you planning to update TeslaPy to support the new Tesla Vehicle Command protocol?
It seems Tesla makes it possible to set up a proxy server to maintain compatibility with Owner API"

It seems like you're hoping someone (possibly the owner of TeslaPy?) would set up a public proxy for the the Owner API and folks could have TeslaPy use it rather than Tesla's servers. That would work, except:

Any proxy uses the tesla-control program to send commands to the vehicle (the proxy will use the old API for some other things).

- Tesla's docs specifically warn not to make the proxy public without some sort of access restrictions as an unsecured proxy might generate lots of requests to Tesla, and Tesla might rate limit requests coming from the proxy.

- In order for a proxy to send commands to a vehicle with the tesla-control program, the owner of the proxy must have registered for FleetAPI access. This means all users of that proxy will appear to Tesla as coming from that registered developer. For now the FleetAPI is free. Tesla has been clear they will start charging registered developers to use the FleetAPI. This makes it unlikely anyone would set up a public proxy server for general/free use.

Another minor wrinkle is that cars will only accept commands from developer apps that the owner of the vehicle has previously approved. That's not a big deal, as the owner of the proxy merely needs to supply a URL that user can click on which will send a request to Tesla to open the owner's Tesla App to approve access. It looks like https://tesla.com/_ak/DEVELOPER.COM where DEVELOPER.COM is replaced by the domain name the registered developer used to gain access to the FleetAPI.

mshoe007, 2024-01-27:
@JanJaapKo "I manually changed the code to fetch the product list iso vehicle list, so it works again. However, when running a new install or upgrade with pip, the version stays at 2.8.0. Any ideas?"

I don't understand. pip install and 'new install' grab tdorssers last published release. Of course the version number does not change.

If you have installed TeslaPy locally, you need to modify your local copy of the python TeslaPy package. Doing pip upgrade or a fresh install could destroy your local changes.

After you make your local change to use the product list, if you want a different version number (which is a good plan), you should find the TeslaPy/__init__.py file on your computer and modify the `__version__ =` line

mshoe007, 2024-01-27:
I wrote

> It seems like you're hoping someone (possibly the owner of TeslaPy?) would set up a public proxy for the the Owner API and folks could have TeslaPy use it rather than Tesla's servers. That would work, except:

It looks like at least one developer is going to do this. See https://github.com/alandtse/tesla/issues/743#issuecomment-1879967335

> I am planning to launch a low cost API only serverless competitor for Home Assistant users: https://teslemetry.com/  Just waiting for Tesla to sign my Fleet Telemetry CSR, but the Fleet API is fully functional.

If that happens, and TeslaPy supports alternative REST servers, then that may be one option to continue using TeslaPy

mshoe007, 2024-01-27:
FYI: This developer has forked the Tesla proxy code to create a proxy that does not require using the FleetAPI

https://github.com/lotharbach/tesla-command-proxy/tree/owner-proxy

@DeLN0 - it still requires someone to set up such a proxy with public access for use by TeslaPy. Or, you can run the proxy locally for yourself.

DeLN0, 2024-01-28:
> FYI: This developer has forked the Tesla proxy code to create a proxy that does not require using the FleetAPI
>
> https://github.com/lotharbach/tesla-command-proxy/tree/owner-proxy
>
> @DeLN0 - it still [...]

@mshoe007 Thats great, ty for that!
Running a proxy server locally seems the best path for now.

Waiting for Tesla to release their paid plans, just hoping prices don't go the way of Twitter API because that would be a bad sign for the open source community.

tdorssers, 2024-01-28:
> @tdorssers I manually changed the code to fetch the product list iso vehicle list, so it works again. However, when running a new install or upgrade with pip, the version stays at 2.8.0. Any ideas? [...]

Published new version on Pypi including vehicle endpoint fix. It does not include the ability to send commands.

tdorssers, 2024-01-28:
> @tdorssers Ty for the commits!
>
> Do you think the Owner API hence TeslaPy will still work in the future (regardless of what Tesla says) ? Because for now the Android app still uses Owner API i [...]

Yes I believe that the Owner API will still work in the future. But we need a Python implementation of the Tesla Vehicle Command Protocol otherwise we will be stuck using a proxy written in the Go language.

HenryRc0, 2024-01-29:
> Published new version on Pypi including vehicle endpoint fix. It does not include the ability to send commands.

Just updated and everything is working again.
Question about "It does not include the ability to send commands."

The commands I use in my script still work:
python3 cli.py -e $account -c START_CHARGE > /dev/null
python3 cli.py -e $account -c CHARGING_AMPS -k charging_amps=$newamps > /dev/null
python3 cli.py -e $account -c STOP_CHARGE > /dev/null

Car is Model S 12/2022

Henry

the-mace, 2024-01-31:
> The writing is on the wall and we have to migrate to the new Fleet Api to ensure we are not left behind.

I haven't looked into the fleet API. I know those like Tezlab, Teslafi etc have moved over. But for individuals, is there a free version? I dont want to pay (a third party or Tesla) for basic API queries about my car (like if its plugged in at night so I dont forget).

DeLN0, 2024-01-31:
> I haven't looked into the fleet API. I know those like Tezlab, Teslafi etc have moved over. But for individuals, is there a free version? I dont want to pay (a third party or Tesla) for basic API [...]

It's free for now but with limits.
See https://developer.tesla.com/docs/fleet-api#membership-tiers

dan-cristian, 2024-02-01:
I updated with the latest changes and it works, both reading and sending commands still works, so good news so far!

DaveTBlake, 2024-02-02:
> have you managed to register? I've failed with "unable to onboard"

@doubledrat I have failed to register too, just many "unable to onboard" when trying to register with my UK business no matter what I did with my website certification and DNS record, or with the scopes (I only want battery stuff but added vehicle just in case).

Did get further by 'accidentally' entering as a US business, but then recieved an email rejection saying `Invalid origin, redirect, or return URI/URL. Update client details and resubmit.` but I don't know if that is upset by my UK account email and SSL combination or if something is still wrong with my website.

My other wonder is if the UK business version of the registration page is broken. Browser console shows error "Failed to load resource: the server responded with a status of 400 (Bad Request)".

I am open to suggestions, is there is a way to get answers from Tesla?

DeLN0, 2024-02-02:
@doubledrat @DaveTBlake 

Yes, I've managed to register but i had to buy a new domain specifically for that.
No clue what the issue is with my old domain, everything was set up correctly.

> Did get further by 'accidentally' entering as a US business, but then recieved an email rejection saying Invalid origin, redirect, or return URI/URL. Update client details and resubmit. but I don't [...]

If you want a key just for personal testing, you can use *any* website and business you want from the internet. 
I have tried and even tesla.com works.
Not recommending to do that and not sure if legal, but it works.

The issue is if you expect to get back a callbak url, for that you have to own the domain.

DaveTBlake, 2024-02-02:
> If you want a key just for personal testing, you can use any website and business you want from the internet.

@DeLN0 really!?!?!? Don't I need to do something with code validation at the web site?  All I want is to fetch data and control _my_ powerwall, I don't care about cars or other people etc. I thought I had to jump through the business  and third-party app hoop to be able to use the Fleet API. I know that the Owner API still works, but for how long, and using a published and offical API seems a better idea, but it is just for my own personal 'hobby' use.

DeLN0, 2024-02-02:
> @DeLN0 really!?!?!? Don't I need to do something with code validation at the web site? All I want is to fetch data and control _my_ powerwall, I don't care about cars or other people etc. I thought [...]

@DaveTBlake 

You don't need to do anything with code validation at the web site, Dave.
The code will be displayed in your browser, you can copy it from there.
You can dm me if you need help.

fabianhu, 2024-02-04:
I goofed around with the developer registration process and made this: [tesla_api_example](https://github.com/fabianhu/tesla_api_example)

For the implementation of the command protocol, we can team up, as I myself would like to have a python-only implementation. @tdorssers Do you have plans?

Meanwhile even the documentation is present, but I'm not feeling motivated enough to code it [myself](https://github.com/fabianhu/tesla_api/blob/eb11d29a82509dfd651122b171c63ff189735600/tesla_api_2024.py#L217). 
For the moment I integrated the vehicle-command Go binary on my Raspberry.

JanJaapKo, 2024-02-04:
@fabianhu  I'm trying your example but I ran into the registration form. Do the account details need to be legit? In other words, do I need a valid VAT number connected to an actual business?

And I notice that you disabled the issues, done that deliberately? Only improvements in the for of PR's allowed ;) ?

fabianhu, 2024-02-04:
Thank you Jan, I merged your improvements. The issues are open in the [tesla_api](https://github.com/fabianhu/tesla_api) repository.

Kamprath, 2024-02-07:
> I managed to get developer access working yesterday following this tutorial.
> 
> Edited to show author's blog instead of the medium.com version which is paywalled:
> 
> https://shankarkumarasamy.blog/2023/10/29/tesla-developer-api-guide-account-setup-app-creation-registration-and-third-party-authentication-configuration-part-1/

This only seems to be available to businesses, which is why I find this change to be so disappointing for the rest of us. Is there any possibility of getting developer access as an indie developer?

JanJaapKo, 2024-02-07:
Yes you can I googled for 'what does a VAT number look like', got a fake number and used my home adress. Worked fine. Check also the improved description at https://github.com/fabianhu/tesla_api_example

mshoe007, 2024-02-07:
> This only seems to be available to businesses

I'm not a business. I signed up for it just as I posted - following that blog. The bogger didn't sign up as a business, either.

I am in USA - I have no idea what the requirements are outside of USA.

Kamprath, 2024-02-07:
> Yes you can I googled for 'what does a VAT number look like', got a fake number and used my home adress. Worked fine. Check also the improved description at https://github.com/fabianhu/tesla_api_example

That is great news - thank you.

gmcluhan, 2024-02-25:
I found this app on the App Store: https://apps.apple.com/us/app/auth-app-for-tesla/id1552058613 and it says it can get API tokens using the Fleet API, once you have that can you swap these tokens into the existing TeslaPy scripts and keep using the scripts already written or did the end points change?