tdorssers/TeslaPy #172: Setting backup_reserve_percent response code 201

Vikings39, 2025-07-24:
Since 19th July 2025 I've been seeing an error when setting the Powerwall backup reserve. The operation seems to make the correct change on the Powerwall but there is an error reported from the code:

  File "/usr/local/lib/python3.13/site-packages/teslapy/__init__.py", line 839, in set_backup_reserve_percent
    return self.command('BACKUP_RESERVE',
           ~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
                        backup_reserve_percent=int(percent))
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/teslapy/__init__.py", line 803, in command
    if response['code'] == 201:
       ~~~~~~~~^^^^^^^^
KeyError: 'code'

AndrewTapp, 2025-07-24:
I too am receiving the KeyError: 'code' error.

I've made the change as you suggested and received a further error...

sensor.py in https://github.com/alandtse/tesla/tree/main produces the error 
`TypeError: type NoneType doesn't define __round__ method`

I've corrected that by changing 
`    @property
    def native_value(self) -> int:
        """Return battery energy remaining."""
        return round(self._energysite.energy_left)`

to
`    @property
    def native_value(self) -> int:
        """Return battery energy remaining."""
        return round(self._energysite.energy_left or 0)`

however I now get 
`Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/automation/__init__.py", line 717, in async_trigger
    return await self.action_script.async_run(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        variables, trigger_context, started_action
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1836, in async_run
    return await asyncio.shield(create_eager_task(run.async_run()))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 460, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 526, in _async_step
    self._handle_exception(
    ~~~~~~~~~~~~~~~~~~~~~~^
        ex, continue_on_error, self._log_exceptions or log_exceptions
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 556, in _handle_exception
    raise exception
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 524, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1012, in _async_step_call_service
    response_data = await self._async_run_long_action(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 624, in _async_run_long_action
    return await long_task
           ^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2799, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2842, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1011, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
        hass, entity, func, data, call.context
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1083, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 122, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/tesla_custom/number.py", line 125, in async_set_native_value
    await self._energysite.set_reserve_percent(value)
  File "/usr/local/lib/python3.13/site-packages/teslajsonpy/energy.py", line 219, in set_reserve_percent
    if data and data["response"]["code"] == 201:
                ~~~~~~~~~~~~~~~~^^^^^^^^
KeyError: 'code'`

which looks to be an issue with teslajsonpy which is the same as the original error, do you have any thoughts?

ayshfm1, 2025-07-24:
I don't think I'm running up to date copies of the software and I'm not going to fix that until the library is updated as it all works but feels fragile.   So my baseline is different.

If I were you I would look at line 219 in File "/usr/local/lib/python3.13/site-packages/teslajsonpy/energy.py"

Put in a print(response) just after the api call likely line 218 and see what you get back.

That said I think it's odds on the same thing code has become Code and the return value is 200 not 201 (and by implication message is now Message).

AndrewTapp, 2025-07-24:
OK, have changed
`    async def set_operation_mode(self, real_mode: str) -> None:
        """Set operation mode of Powerwall.

        Mode: "self_consumption", "backup", "autonomous"
        """
        data = await self._send_command(
            "BATTERY_OPERATION_MODE",
            path_vars={"site_id": self.energysite_id},
            default_real_mode=real_mode,
        )
        if data and data["response"]["code"] == 201:
            self._site_config.update({"operation": real_mode})

    async def set_reserve_percent(self, value: int) -> None:
        """Set reserve percentage of Powerwall.

        Value: 0-100
        """
        data = await self._send_command(
            "BACKUP_RESERVE",
            path_vars={"site_id": self.energysite_id},
            backup_reserve_percent=int(value),
        )
        if data and data["response"]["code"] == 201:
            self._site_config.update({"backup_reserve_percent": value})`

to
`    async def set_operation_mode(self, real_mode: str) -> None:
        """Set operation mode of Powerwall.

        Mode: "self_consumption", "backup", "autonomous"
        """
        data = await self._send_command(
            "BATTERY_OPERATION_MODE",
            path_vars={"site_id": self.energysite_id},
            default_real_mode=real_mode,
        )
        if data and data["response"]["Code"] == 200:
            self._site_config.update({"operation": real_mode})

    async def set_reserve_percent(self, value: int) -> None:
        """Set reserve percentage of Powerwall.

        Value: 0-100
        """
        data = await self._send_command(
            "BACKUP_RESERVE",
            path_vars={"site_id": self.energysite_id},
            backup_reserve_percent=int(value),
        )
        if data and data["response"]["Code"] == 200:
            self._site_config.update({"backup_reserve_percent": value})`

in energy.py from https://github.com/zabuldon/teslajsonpy 
and added the package as a custom component and all currently looks good, many thanks for your help as far!!

MikeScavell, 2025-07-24:
After nothing some discussion here and back in the pypowerwall repo, I made a change to the pull request to handle both the 200 and 201 response. As others have stated, the response seems to differ between set-mode and set-reserve now for some reason. Anyway, what I have appears to be a working solution for this repo, but happy to take feedback for changes (or feel free to discard my solution if there is a better way to handle it).

https://github.com/tdorssers/TeslaPy/pull/171

cjw296, 2026-02-17:
@tdorssers - are you able to do a release any time soon?